OpenIPC/majestic-webui #20: The new user from the ONVIF guide has administrative access (security issue)

MikeVRT, 2024-09-13:
_I'm not sure about right choice of sub-project. Please move it to firmware if it fits better there._

The user, created with that [guide](https://github.com/OpenIPC/wiki/blob/master/en/majestic-streamer.md#onvif) for ONVIF has WebUI administrative access.

The simple solution, from my point of view, is:
It may be better to allow only users in a special system group to log in to the WebUI.
For example, the root user is in the **wheel** group by default:
  _uid=0(root) gid=0(root) groups=0(root),10(wheel)_
This 'marker' could be easily used as a condition to determine whether a user could access the system as root or just as an endpoint consumer. Just a simple check in the WebUI authentication code and an 'Access Denied' message if the user is not a member of the "right group".

flyrouter, 2025-02-25:
Thank you for your message and suggestion.

At the moment the system is single-user but we have made a workaround that currently does not allow users other than "root" to enter the interface, but they can receive media content (jpeg, mjpeg, rtsp) and also execute .cgi scripts from the "/v" directory, control the backlight and some other actions of a regular user or an automated system.

This ticket will be closed after the information is transferred to the Wiki with explanations.

MikeVRT, 2025-04-04:
> but they can receive media content (jpeg, mjpeg, rtsp)

I found an issue: the endpoint 'image.jpg' works only under root authorization.
This is the actual version for today.(2.4.03.31).

MikeVRT, 2025-05-20:
The bug still here - v.2.5.05.19
Endpoints like image.jpg are not accessible via non-root credentials in the link. Only non-root links to the RTSP streams still work.