OpenIPC/majestic-webui #30: Backup creation always fails with 500 Internal Server Error

qumuq-til, 2025-02-25:
/cgi-bin/ext-backuper.cgi?backup=create accessible from Firmware - Settings - Create Backup always fails.

del000, 2025-02-25:
> Good day You can request research and correction directly from the author of the solution - [@del000](https://github.com/del000)

@qumuq-til tnanx for ur feedback!

pls, try temporary workaround from camera`s console: 
```
chmod +x /var/www/cgi-bin/ext-backuper.cgi
```
and after this, try to backup from web

@flyrouter i dont have any ideas, how to fix exec bit on file in repo, can u show me the way?

qumuq-til, 2025-02-25:
@flyrouter I don't see the point of assigning me here when @del000 is clearly the developer responsible for the module in question? Could you please remove this assignment?

del000, 2025-02-25:
@flyrouter  fixed, approve the PR, pls: https://github.com/OpenIPC/majestic-webui/pull/32

flyrouter, 2025-02-25:
> [@flyrouter](https://github.com/flyrouter) i dont have any ideas, how to fix exec bit on file in repo, can u show me the way?

Hi.

Thanks, the command check worked, everything is fine.

I think you need to give rights to the local file and make a PR, this is the easiest way.

flyrouter, 2025-02-25:
> [@flyrouter](https://github.com/flyrouter) fixed, approve the PR, pls: [#32](https://github.com/OpenIPC/majestic-webui/pull/32)

Done.

Thank you very much for the promptness.
I am closing the Issue as the nightly build will do all the work.

Thank you all.

closed 2025-02-25 by flyrouter