OpenIPC/majestic-webui #31: Make web interface accessible without internet

qumuq-til, 2025-02-25:
Currently it's not possible to access Majestic web front end without an active internet connection because it's stalling on googleapis.com request, seemingly trying to download fonts or css or similar data. It must always be possible to access the interface locally without any internet connection.

qumuq-til, 2025-02-25:
Please remove this assignment. I merely reported a bug, I'm not associated with the OpenIPC project and have no intention to contribute at this moment.

flyrouter, 2025-02-25:
Ok, I understand you. The Issue attachment has been removed.

Usually when writing about a problem, it is assumed that it will first be studied and suggestions or wishes will be given in an expanded form, and not just a note "does not work".

Our firmware is a network camera that connects to the Internet and, accordingly, some components can be loaded. If the components do not load or you need to use the camera autonomously, you can either enable options in the browser or send a patch or use the API and CLI to configure without a web interface.

Thus, if you want this ticket to be processed, please let us know which component you cannot load and for what reason. The current web interface considers some components optional, but your browser may think differently. Therefore, only extended teamwork is always needed.

I hope for your understanding and have a nice day.

qumuq-til, 2025-02-27:
Sure. So the problematic bit was the
```
@import url('https://fonts.googleapis.com/css2?family=Montserrat:wght@400;500;700&display=swap');
@import url('https://fonts.googleapis.com/css2?family=PT+Mono&display=swap');
```
This pulls in CSS files that in turn request https://fonts.googleapis.com/...

I may have hit an edge case, actually, but it could still be important. I'm using a VPN solution that creates new routes in the routing table. After a suspend/resume cycle that solution doesn't automatically refresh its connection, so the route via its virtual interface persists and any request to any internet host will stall forever without timing out (as would normally be the case if the network is fully down). That is why my browser is stalling on trying to request googleapis.com and the contents of Majestic UI are never shown. 

I think it might be a good idea to have those fonts locally in the firmware itself. Those are tiny in size, and in my view, the less requests a device makes to the internet, exposing itself in the process - the better. Take for example how the update checks are handled currently - only when you browse to the 'Update' page, a request to GitHub is being made. It doesn't happen, for example, on the main Status page, and I think it's absolutely the right way to do it.

Privacy Badger extension also doesn't seem to like fonts.gstatic.com, blocking cookies coming from this domain when a Majestic page with monospaced fonts is being shown, which is another concerning point.

closed 2026-08-29 by openipc-ai