OpenIPC/majestic-webui #45: Allow inverting backlight pin

jangrewe, 2025-07-12:
Hi, i have a `hi3516ev300` board that's i've just flashed and now i'm trying to figure out all the GPIO pins... 
I found both pins for IR Cut (11 & 10) and also the pin for enabling the floodlight, which is 72 - but unfortunately it's inverted, so the floodlight is always on. 

Could you please also add a switch to invert "GPIO pin for camera light" (`_nightMode_backlightPin`)?

Thanks!

flyrouter, 2025-09-13:
Good day
We invert only incoming signals, which can be different.
In your case, you can simply swap the numbers locally and everything should work as planned.
It is impossible to provide for the correctness of the IR pins and their compliance, since the hardware is assembled by different factories in China.

jangrewe, 2025-09-13:
Oh, that's a simple solution - thanks for the hint!